mriscoc/Ender3V2S1 #1423: [BUG] Laser module support causes M104 to stop the hotend fan momentarily

Ran9, 2025-03-01:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

In firmware versions with support for the laser module, M104 command stops the hotend cooling fan momentarily.
Tested in a custom build and the precompiled firmware from [https://github.com/mriscoc/Special_Configurations/releases/tag/CVLM](url) specifically Ender3S1-F4-UBL-CVLM-20240127.bin

This does not seem to happen in precompiled versions that don't have laser module support.
I searched in the source code but could not find the cause.

### Expected behavior

M104 command should set target temp without affecting the fan.

### Steps to Reproduce

1. Flash firmware from [https://github.com/mriscoc/Special_Configurations/releases/tag/CVLM](url)
2. Issue command M104 S100
3. Wait for hotend fan to turn on
4. Issue M104 again, for example M104 S0
5. Hotend fan turns off for a second and then back on again

### Version of Professional Firmware

Ender3S1-F4-UBL-CVLM-20240127

### Printer model

Ender 3 S1

### Electronics

Board with STM32F401RCT6

### LCD/Controller

DACAI

### Bed Leveling

None

### Your Slicer

Prusa Slicer

### Host Software

None

Ran9, 2025-03-19:
Looks like it is caused by this code in /Marlin/src/gcode/temp/M104_M109.cpp:
`TERN_(CV_LASER_MODULE, laserOn(false));`
`// #if ENABLED(CV_LASER_MODULE)`
`//   if(laser_device.is_laser_device()) return;`
`// #endif`

The same code exists in M140_M190.cpp so M140 command is affected as well.

Something in the function `laser_set ` in cv_laser_module.h affects the hotend fan, though the source code for that function doesn't seem to be available.

mriscoc, 2025-03-19:
The CVLM version is for use with the Creality CV Laser module: https://www.creality.com/es/products/cv-laser-module
This module is connected to the to the same pin like the hotend part fan.

https://github.com/mriscoc/Special_Configurations/releases/tag/CVLM

Ran9, 2025-03-19:
I do have the laser module and it works great with the pro firmware.
The issue was fixed by commenting out the first line and un-commenting the following lines in the code I mentoined previously.

TheWebMachine, 2025-03-19:
Yes, this is normal behavior and has existed since laser module support was added as a special firmware version. I never reported it as a "bug" because I know how the laser is wired up (for the uninitiated, I am the sponsor of the CVLM variant of this firmware), and how that relates to the normal hotend. I do use both heads on my printer without issue, simply toggling the menu option when switching toolheads. The momentary pause of the fan on the extruder heat break is less than a second and has proven to cause no issues in practical use, no matter which filament I print with. So long as the heat break fan has no blockages from dust and such, it has more than enough airflow to keep the heatsink well below temps, at least in my testing. The momentary interruption isn't long enough to saturate the heatsink.

While it *might* be possible to somehow tie execution of the above code to whether or not the Laser Module option is enabled, I don't know that it is explicitly worth the effort if it could break something else in the process. Getting laser working **well** alongside normal printing was no easy task, given the limited flash space on these F4 boards.

-----
As an aside, my apologies @mriscoc! I know I've been absent a while. I actually ended up completely rebuilding my print lab over the winter due to simply outgrowing the previous room in my house. 🤣 That said, I'm back up and running now and will be testing the latest CVLM version soon to see how it compares to...{boots printer}...20230904. 😆😎

mriscoc, 2025-05-19:
Any news about this issue? is it a false alarm?

TheWebMachine, 2025-05-19:
> Any news about this issue? is it a false alarm?

I knew there was something I forgot to do! 🤦‍♂️ I have been printing and engraving for the last couple of weeks and completely forgot to install the latest firmware before I started firing stuff off the bed! 🤣

I was just about to start another engraving run when I saw your notification, so I'll load up the latest FW, test the laser since it is currently attached, then switch back to the hotend for a quick print to confirm all is well. 

That said, I don't think the above issue will turn out to be anything worth tackling, if it *is* the same issue I noted 18 months ago and wrote off as inconsequential/burdensome to fix. I can't find my original comment on the matter at this point because we had so much discussion in the early stages of building out laser support, but I do distinctly remember we talked about it and decided to leave it alone at the time. 

Of course, I will thoroughly test it out and see if we need to revisit it as a real problem. I'll report my findings over the next couple of days as I complete some jobs.

Thanks for the reminder! 🙃

mriscoc, 2025-05-19:
> That said, I don't think the above issue will turn out to be anything worth tackling

Thanks for your confirmation.

closed 2025-05-19 by mriscoc