mriscoc/Ender3V2S1 #1432: Z wizard changes Z probe setting between prints

MrFarporte, 2025-03-17:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

It forgets the Z offset setting.  This has been described in a previous bug report, but it is still happening with the newest software on my Ender 3 V2 with BL touch and spider hotend.  I know the offset is 3.33, but it changes the nozzle height to 0.5 too high when beginning the print.  I can go back at change the Z offset setting up and down and set it back to 3.33 and it prints fine.  Once I would tell it to Z home in the Z probe wizard, and it would got up several MM in height each time, instead of going down to the offset height I set.  I had to turn off the power and hard reset it.  

### Bug Timeline

It started when I install the BL touch 2 weeks ago.

### Expected behavior

I expected the Z probe to stay at the same offset from the nozzle height after doing the paper slip manual calibration.  

### Actual behavior

It would start printing too high, or not set the Z offset correctly in Z wizard. I would reset the printer to get it to print correctly.

### Steps to Reproduce

1.  Install a BL touch
2.  Install the software
3. Go to Z wizard
4. Set Z offset of nozzle height
5. print

### Version of Professional Firmware

 Ender3V2-422-BLTUBL-MPC-20240125

### Printer model

Ender 3 V2

### Electronics

4.2.2

### LCD/Controller

Dwin

### Other add-ons

BL Touch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

woodworker, 2025-03-19:
I don't know if this helps, but did you also saved the settings before print?

mriscoc, 2025-03-19:
From the last versions, the Z offset is not saved automatically to be compliant with AML 2, you need to save settings manually.

MrFarporte, 2025-03-23:
Here is what happened this time.  I turned the printer off.  I turned it back on and did the bed tram with pre-heated hotend and bed. I have the direct drive factory setup, with the metal hotend cover, a spider hotend, and the BL touch kit.  I then did an automesh bed level. I then went to Z wizard and re-Set it to -3.33 with a piece of paper.   I saved the setting with "store settings".  I successfully printed 2 prints about 3 hours each.  The bed stayed tight and secure, the bed level remained level after automesh leveling again.  I turned the printer off.  I turned it back on to print a 3rd print, and did the re-level but not Z probe wizard check.  The print started the 1st layer too high again.  I stopped the print, and re-heated the bed and hotend, and checked the Z setting with a piece of paper, and it changed to -3.59.  isn't this the offset from the BL touch probe to the hotend nozzle tip?  Why would this change?

LucasTolosa, 2025-04-17:
Hi, I’m experiencing exactly the same issue with the Mriscoc firmware — both the 2024 and 2025 versions, and even with Jyers!  
But I was able to fix it with an easy workaround.

Maybe I’m over-explaining the first part, but it might help someone :)

**Pre-condition:**  

1. Run auto home on all axes.  
2. Manually level the bed at *at least* 5 points using the paper method.  
3. Then adjust the Z offset: move the Z axis to position 0, and fine-tune the Z offset to the optimal value — again using the paper.  
4. Now, run **Auto Home on the Z axis**, then **Auto Home on all axes** (it’s essential to do *both* and in that *order*).  
5. Immediately generate the mesh and save the changes.  
6. Check that everything is correct: move the Z axis to position 0 and make sure it’s just as you previously adjusted.  If everything looks good, we can say it's correctly calibrated.

**The problem:**  
In some cases, the printer bugs out and the real Z-axis position becomes offset by 5mm (in my case).  
If I tried to print, it would print in the air, and if I tried to correct it, it could damage the bed.

**Solution to prevent misalignment:**  
To recover or make sure the Z axis is in the calibrated position, first run auto home **only on the Z axis**, and then auto home **on all axes**.  
I tested this many times and it worked — even after manually moving the Z axis with the motors disabled.

**Improvement:**  
To automate this, simply add it to the slicer’s start G-code so it runs before every print:

```
...
G28 Z ; Home only Z axis  
G28   ; Home all axes
...
```

(Sorry for my English, I used Google Translate)

LucasTolosa, 2025-04-17:
After countless attempts and a lot of patience, I finally found a way to calibrate it properly — and make sure it stays calibrated.

Initial setup
I start with my Z offset set to 0, unconfigured.

**let's level the bed**:
- Go to Homing > Auto Home. (At this point, it tells me that the Z axis is at position 7.55 — but this isn’t accurate. If I move it to position 0, it’s still far from the bed. If I try to correct it manually, the bed can get damaged during a print because the position displayed is not the real one. So, ignore the Z position shown here, it’s not correct.)
- After running Auto Home, Go to Homing > Home Z.
- Then go to Bed Tramming, choose a point (e.g., Front Left). Adjust the bed springs so that the displayed Z position is close to 0. Repeat this for all tramming points until they’re all near zero. Now the bed is leveled.

**Determining the correct Z Offset**
The bug is that the Z position is not represented accurately.
To get the real position, follow these steps exactly:

- Homing > Auto Home
- Homing > Home Z
- Homing > Auto Home (again). Now the printer is at the real Z = 5 position.
- Move Axis > Move Z to 0
- Z Probe Wizard > Probe Z Offset: Start from Z offset = 0, and use a piece of paper to bring the nozzle close to the bed. For example. in **my case**, the correct Z offset was -2.50.
- Z Probe Wizard > Store Settings

Final step
Now it’s calibrated — but it could lose calibration again.
To prevent this from happening, you can either:

Manually repeat:
Auto Home → Home Z → Auto Home before printing

Or automate it by adding the following to your slicer’s start G-code:

```
G28 Z ; Home only Z axis  
G28   ; Home all axes
```"

its works for me. I hope this helps someone, and hopefully this can be fixed in a future version since it was very frustrating to find the solution.

MrFarporte, 2025-04-20:
> After countless attempts and a lot of patience, I finally found a way to calibrate it properly — and make sure it stays calibrated.
....
> Manually repeat: Auto Home → Home Z → Auto Home before printing
> 
> Or automate it by adding the following to your slicer’s start G-code:
> 
> ```
> G28 ; home all axis
> G28 Z ; Home only Z axis
> G28 ; home all axis
> ```
> 
> its works for me. I hope this helps someone, and hopefully this can be fixed in a future version since it was very frustrating to find the solution.

Yes! It worked.  I tried the manual method first. Saved the settings.  Now it prints correctly.  The Z Probe setting up at -3.36, which is interesting because it's right in between -3.33 and -3.39 which were the 2 settings it was bouncing between before.  I now put the addition G28 lines in my slicer to add to every gcode file.  

Thanks!

mriscoc, 2025-04-26:
For reference, printers that have this type of issue probably have a Z-backlash problem.
In those printers, the Z-offset will change between two different values depending of the last Z movement. This issue can be corrected by mechanical adjusting the Z axis or using a software compensation like the [M425](https://marlinfw.org/docs/gcode/M425.html) (This G-code is not enabled by default)

MrFarporte, 2025-05-03:
This could be a false alarm from z-axis movement.  I think there is a z-binding or z-backlash issue caused by my hardware.  I do notice that the stepper motor gets hot and isn't turning the z-screw up enough.  I have attempted to adjust the z-screw angle, but these ender 3 series are prone to z hardware issues do to the mounting point design tolerances for the z-screw.  The print makes it to about 1mm or 3mm and the z height is too low. the nozzle either clogs and the print is smashed into the bed, or the print is pushed off of the bed.  either way, it sucks.

closed 2025-07-13 by github-actions[bot]