mriscoc/Ender3V2S1 #1448: [BUG] unified_bed_leveling.cpp is missing

furrywall, 2025-05-11:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

When I was compiling the firmware and activated UBL I found that the file unified_bed_leveling.cpp from the modules folder was missing. 

### Expected behavior

I expected it to compile 

### Actual behavior

It did not compile. I showed me this: 

c:/users/kiero/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: Marlin/lib/proui\libproui.a(proui_ex.cpp.o): in function `MarlinSettings::store_mesh(signed char)':
9][9])'
c:/users/kiero/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: Marlin/lib/proui\libproui.a(proui_ex.cpp.o): in function `MarlinSettings::load_mesh(signed char, void*)':
(.text._ZN14MarlinSettings9load_meshEaPv+0x6e): undefined reference to `unified_bed_leveling::set_mesh_from_store(short const (&) [9][9], float (&) [9][9])'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\STM32F103RE_creality_xfer\firmware.elf] Error 1

### Version of Professional Firmware

2.1.3 

### Printer model

Ender 3 V2

### Electronics

4.2.2

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

OctoPrint

### Additional information & file uploads

I fixed it myself by adding the file with:

#include "../inc/MarlinConfigPre.h"
#include "../inc/MarlinConfig.h"

namespace unified_bed_leveling {

  // Convert stored Z-values into float mesh values (from short to float)
  void set_mesh_from_store(const int16_t (&z_values)[9][9], float (&mesh)[9][9]) {
    for (uint8_t x = 0; x < 9; x++) {
      for (uint8_t y = 0; y < 9; y++) {
        mesh[x][y] = z_values[x][y] * 0.01f;  // Convert stored values to float mm
      }
    }
  }

  // Convert float mesh values into storable Z-values (from float to short)
  void set_store_from_mesh(const float (&mesh)[9][9], int16_t (&z_values)[9][9]) {
    for (uint8_t x = 0; x < 9; x++) {
      for (uint8_t y = 0; y < 9; y++) {
        z_values[x][y] = static_cast<int16_t>(mesh[x][y] * 100.0f);  // Convert float mm to stored int
      }
    }
  }

}

And also I added to platformio.ino with : 

+<src/module/unified_bed_leveling.cpp>

mriscoc, 2025-05-12:
You need to download the proper versions  (2024 or 2025) of the two repositories, the main one and the Special_Configurations: https://github.com/mriscoc/Special_Configurations#creating-your-own-custom-configuration

classicrocker883, 2025-05-30:
I think you need `OPTIMIZED_MESH_STORAGE` enabled

closed 2025-08-10 by github-actions[bot]