mriscoc/Ender3V2S1 #1449: [WIKI] Missing images - versions page

TheWebMachine, 2025-05-20:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Just wanted to let you know that a few images have vanished from the following Wiki page:

https://github.com/mriscoc/Ender3V2S1/wiki#versions

### Version of Professional Firmware

N/A

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

mriscoc, 2025-05-20:
Please change your browser, I test with Chrome and Edge and I can see all the images.

TheWebMachine, 2025-05-20:
Actually, I think it was a Github CDN issue. They didn't load for over an hour as I was bouncing around the git ( I checked back 3 times before posting), but now they loaded just fine...same browser, PC, internet connection. {shrugs}

closed 2025-05-20 by TheWebMachine