mriscoc/Ender3V2S1 #1450: Updating TJC firmware on mac

lukeaweis, 2025-05-21:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

When trying to update the firmware on a TJC screen, I added the TJC_SET folder to an empty, properly formatted sd card. It appears to only have the tjc.tft file in it, but when I put the SD card in the screen and turn the printer on, it won't update because of "multiple tft files" 

I figured out that MacOS adds metatdata files including "._tjc.tft". You can see these hidden files using ls -a in terminal. You can run rm rm "._tjc.tft" to fix this and then the card updates the screen just fine. 

### Expected behavior

I expected the firmware to properly update. 

### Actual behavior

The screen listed the error "Baud: 115200 SD Card Update... The system detects multiple tft files, can not be upgraded, please remove the extra tft file"

### Steps to Reproduce

formatted new sd card
downloaded TJC_SET folder and moved it to the sd card
inserted the sd card into the back of the lcd and powered on the printer

### Version of Professional Firmware

Ender3V2-422-BLTUBL-TJC-MPC-20240127.bin

### Printer model

Ender 3 v2 Neo

### Electronics

4.2.2

### LCD/Controller

TJC

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

lukeaweis, 2025-05-21:
I found the solution in this [discussion](https://forum.arduino.cc/t/nextion-touch-display-more-than-one-tft-file-on-sd-card/417411/8) but if you could add it to this guide it would help thanks

mriscoc, 2025-05-21:
Please use the section [Discussions](https://github.com/mriscoc/Ender3V2S1/discussions).
This Issue Queue is for firmware bug reports and development-related issues, and we prefer not to handle user-support questions here.

closed 2025-05-21 by mriscoc

mriscoc, 2025-05-21:
Discussion related to this issue: https://github.com/mriscoc/Ender3V2S1/discussions/1451