mriscoc/Ender3V2S1 #1455: [BUG] Bigtreetech skr mini Bl touch intermittently doesnt trigger

balt001, 2025-05-27:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Ender 3 v2 with antclabs Bl touch v3.1 will intermittently not trigger cause nozzle to crash into bed. Mostly doesnt it while building a mesh although is has happened a couple times while tramming and tilting the mesh. It is very random fails about 1 in 20 times. I am 100% certain is it not a loose connect or faulty bl touch have tried different cables and a different bl touch. have read posts where this is a pretty common this however i dont see any kind of solution for it i have video on the probing failing if need to see. Have tried to turn off step smoothing and problem still persists. have tried isolating the cable outside of the main wire loom, different firmware versions including precompiled versions. running out of ideas on what to do.  

### Bug Timeline

first started when i install the btt skr mini 

### Expected behavior

expect it to probe the bed and read the correct results. 

### Actual behavior

it probes, the bl touch flashes like it triggers however the z axis still keeps going down and crashing the nozzle into the bed causing the probe to flash and the screen reads NAN on the probing point of the mesh. 

### Steps to Reproduce

1. preheat bed and nozzle 
2. tram
3. build mesh. 

### Version of Professional Firmware

Experimental 20250111, 20240122, (custom config), precompiled 2025, precompiled 20240122

### Printer model

ender 3 v2, dual z kit, antclabs v3.1 bl touch

### Electronics

skr mini e3 v3 

### LCD/Controller

factory creality ender 3 v2 lcd screen with btt lcd cable. and tried a custom lcd cable.

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

### Additional information & file uploads

Linear advance, input shaping, MPC, step smoothing off. bl touch installed into the 5 pin probe port.

balt001, 2025-05-28:
> Finally someone else has EXACTLY the same problem as I do! This is SO intermittent. I have tried changing the cables, various options in the firmware, but nothing helps. 90% of the time everything works perfectly. Then, suddenly it will crash into the bed once or twice in a row. I have simply got used to hovering my finger over the power button to save mashing the probe into the bed too much. The exact same Bl-touch and cable worked fine on the stock Ender 3V2 Motherboard but this board doesn't support linear advance.

Seems to b an issue with the stm3232g0 that btt used im trying to figure it out between force_sw_mode, adaptive_step_smoothing, messing with irq priorities and different bl touch settings. I compiled the latest marlin and didnt see any probing fails but was only on a small grid mesh. I broke my skr mini somehow the probe pins broke so have a new one coming. Will keep everyone posted. But it seems to be a widespread issues all across github.

balt001, 2025-05-29:
I have still having issues.

balt001, 2025-05-29:
Having really good success with force_sw_mode on , bl touch delay 500, irq priorities <1 (ini<stm32g0), adaptive_step_smoothing off. Not sure if the irq priorities did anything or which feature caused it to work so I will be undoing each one so I can identify which option fixed it. Have created about 20 meshes with fans and heaters on/off and haven't had a single probing fail.

AndySmithers, 2025-06-05:
Well, like balt001, I haven't had a single bed crash or BLT probe failure since implementing these changes.
I have, however, had 1 instance of an inaccurate BLT probing (print head ended up too close to the bed by about 0.5mm).
But I have not been able to replicate this so it may have just been some other gremlin.

closed 2025-08-16 by github-actions[bot]