mriscoc/Ender3V2S1 #1480: [BUG] (bug summary)

SparkyNZ, 2025-07-01:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Nozzle Drops to Bed and Homing X/Y/Z is lost

### Bug Timeline

Started since I installed Experimental-Ender3V2-422-BLTUBL-LA-MPC-20250106.bin

### Expected behavior

I have 2 issues, I don't know if they are related.
1. After Auto Homing, I expect the Nozzle to remain wherever it is placed using Move Z.
2. When I start to print, I expect the nozzle to be at the 0.2mm (or whatever) above the bed

### Actual behavior

I have 2 issues, I don't know if they are related.
1. I do the Auto Homing and use Move Z to place the Nozzle up above the bed somewhere. After several minutes the nozzle drops down to the bed and the X/Y/Z values begin to flash. Homing information is lost for some reason.

2. When I did get printing to start, the nozzle remained about 4-5mm above the bed and filament is dropped from a great height.

My first issue is the biggest concern for me - I don't know if I have a bug with my hardware or the firmware. Could there be a fault with my motherboard that's causing configuration to be lost, or is this something that has been reported before?

### Version of Professional Firmware

Experimental-Ender3V2-422-BLTUBL-LA-MPC-20250106.bin

### Printer model

Creality Ender 3 V2 Neo

### Electronics

4.22

### LCD/Controller

TJC

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

None

SparkyNZ, 2025-07-04:
I got in touch with Creality Support. They recommended I revert back to factory firmware: Ender-3 V2 Neo-Marlin2.0.8.3-HW-V4.2.2-SW-V1.1.6-CR-Touch-202209013A1.bin (and the original TJC_SET display firmware).

That stopped the 2 minute nozzle drop from happening - so does this confirm that there is a bug in the Professional Firmware?

The problem for me is that I need the Professional Firmware at this site because I replaced the hot end and the Creality factory firmware is not capable of recalibrating the nozzle/PID/thermistor behaviour. The nozzle temperature goes up and down and won't start printing with the factory firmware.

I will reinstall the latest Professional Firmware again. Please let me know if there is any way I can retrieve some debug for this issue - even if I do have to recompile the firmware myself.

mriscoc, 2025-07-05:
Disabling the stepper motor drivers is a design feature that can be configured using G-Code M18 and M84.

https://github.com/mriscoc/Ender3V2S1/blob/17d0a29146093f2ca443d15d4bbe42c8db9cb897/Marlin/Configuration_adv.h#L1249-L1264

This is not a general printer support forum. Many problems can be easily solved by reading the firmware [Wiki](https://github.com/mriscoc/Ender3V2S1/wiki#main-topics) please check the calibration guide.

This Issue Queue is for firmware bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

Enhanced support in Patreon: https://www.patreon.com/mriscoc

Free community support:
- Github discussions: https://github.com/mriscoc/Ender3V2S1/discussions
- Facebook: https://www.facebook.com/groups/professionalfirmware
- Telegram: https://t.me/ender3v2s1firmware
- Reddit: https://www.reddit.com/r/Ender3v2Firmware
- Discord: https://discord.gg/3dprinters

closed 2025-07-05 by mriscoc