mriscoc/Ender3V2S1 #1491: [BUG] BLTouch error happens quite often

ernestp, 2025-08-02:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

[BUG] BLTouch error happens quite often on [v2025-01-06](https://github.com/mriscoc/Ender3V2S1/releases/tag/20250106)

### Bug Timeline

Started happen after update from 2024 release

### Steps to Reproduce

1. Preheat PLA
2. Start print with Octoprint
3. Printer panic

### Version of Professional Firmware

 Experimental-Ender3V2-422-BLTUBL-LA-MPC-20250106.bin

### Printer model

Ender 3V2 

### Electronics

4.2.2, BLTouch

### LCD/Controller

DWIN

### Other add-ons

None

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

```gcode
Recv: ok N10 P15 B15
Send: N11 G92 E0*119
Recv: X:0.00 Y:231.00 Z:10.48 E:0.00 Count X:0 Y:18480 Z:4192
Recv: ok N11 P15 B15
Send: N12 G28*32
Recv:  T:219.08 /220.00 B:65.00 /65.00 @:79 B@:48
Recv: echo:busy: processing
Recv:  T:219.05 /220.00 B:65.02 /65.00 @:79 B@:42
Recv: echo:busy: processing
Recv:  T:218.97 /220.00 B:65.04 /65.00 @:80 B@:35
Recv: echo:busy: processing
Recv:  T:218.98 /220.00 B:65.01 /65.00 @:80 B@:44
Recv: echo:busy: processing
Recv: Error:!! STOP called because of BLTouch error - restart with M999
```

mriscoc, 2025-08-06:
This is an M112, it was sent by Octoprint, you need to check the Octoprint topic in the wiki.

ernestp, 2025-08-09:
I was not able to find specific scenario to reproduce issue and it seems after settings reset it happens only once.

I will watch it and update if I find a way to reproduce issue. Meanwhile we can close this issue.

closed 2025-08-10 by ernestp